microscope-cockpit/cockpit #778: Ubuntu 20LTS OpenGL invalid context when acquiring image

juliomateoslangerak, 2021-10-05:
Acquiring an image on Ubuntu 20.04 LTS fails with the following error.
Up to date installation of Ubuntu 20.04LTS. Installation through pip with compilation of wxPython. I cannot use the default OS wxpython cause it provides version 4.0. Cockpit with dummies.
Any idea?

```python
Traceback (most recent call last):
  File "/home/julio/PycharmProjects/cockpit_mri/cockpit/gui/imageViewer/viewCanvas.py", line 627, in onPaint
    self.image.draw(pan=(self.panX, self.panY), zoom=self.zoom)
  File "/home/julio/PycharmProjects/cockpit_mri/cockpit/gui/imageViewer/viewCanvas.py", line 318, in draw
    glVertexPointerf( [(-hlim + i*dx, -vlim + jj*dy),
  File "/home/julio/PycharmProjects/cockpit_mri/venvs/venv38/lib/python3.8/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
  File "/home/julio/PycharmProjects/cockpit_mri/venvs/venv38/lib/python3.8/site-packages/OpenGL/wrapper.py", line 626, in wrapperCall
    storeValues(
  File "/home/julio/PycharmProjects/cockpit_mri/venvs/venv38/lib/python3.8/site-packages/OpenGL/arrays/arrayhelpers.py", line 156, in __call__
    contextdata.setValue( self.constant, pyArgs[self.pointerIndex] )
  File "/home/julio/PycharmProjects/cockpit_mri/venvs/venv38/lib/python3.8/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
  File "/home/julio/PycharmProjects/cockpit_mri/venvs/venv38/lib/python3.8/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context
```

carandraug, 2022-01-11:
This is duplicated from #739 . Closing as duplicated.

closed 2022-01-11 by carandraug

juliomateoslangerak, 2022-01-12:
For the record, indeed as @carandraug said installing 4.1.0 fixes this.