microscope-cockpit/cockpit #783: HW and OS recomendations

juliomateoslangerak, 2021-10-15:
The PC running cockpit is dying (not bad after 8 years) and have to go shopping. I was wondering if there are some key aspects to be aware of concerning HW and software (mainly OS).
The intent of the PC is running as main acquisition workstation. Fast timelapses (sCMOS cameras) in multi-site. Lets assume for the moment that price is not an issue if it provides an advantage.
Here are some thoughts. Please add / comment
### HARDWARE
- PC vs Mac: I assume it is saver (and cheaper) to go PC than AppleM1
- CPU:
  - My guess is that a Dual socket PC would be useless. 
  - For number of cores vs MHz, I would favour MHz as most CPUs would have a good nr of cores...
  - Intel vs AMD. AMDs seem to be better than intels... any reason for getting or not AMD...
- RAM:
  - I believe this might be more dictated by the software that you want to run on the PC for reconstruction, deconvolution, ... 64-128Gb would seem fine for me.
- GPU:
  - What is going to be prime feature for GPUs. I would think that fancy gaming features like ray tracing are not key. I would say amount of Memory might be limiting for the mosaics.
  - Again depending on other software you might be willing to run on top of cockpit.
  - It finally seems that these days you get what you can get.
- Storage:
  - Big and fast, right: I currently have 4xSSDs in RAID0 through a dedicated RAID PCI card. Is this an overkill with nowadays SSD's speed and size?
- Network:
  - I currently have the cameras connected through 10Gb to the switch and the switch to the PC with two 10Gb cables to a dual slot network card that you can, in principle, pool to have 20Gb. Not pooled cause in windows seems to be more difficult. If you do the maths, I believe this might be necessary.
### OS
- MacOS: 
  - Nice (I'm a mac lover) but a config like the one above is going to be overpriced.
- Windows: 
  - I hate windows but users might prefer it to linux (and our IT too). 
  - What about Windows 11, did someone try it yet?
- Linux: Would be my preferred choice. 
  - Seems to be much faster in the tests that you performed for the paper, or is this only in the case of a local network.
  - I have trouble installing Cockpit in Fedora and Ubuntu #777 #778. 
  - Do you guys have preferred distro? I guess I would go for Ubuntu. Why? it feels a better choice for support and user friendliness.
- Dual boot?

iandobbie, 2021-11-09:
1) The OS choice might be down to driver availability for hardware, although if you are using remote hardware then this shouldn't matter. 

2) The interface looks better on windows and Linux than macOS as the main window button heights are ignored. 

3) GPU: the most important thing is to have enough GPU ram to store a realistic mosaic as filling the GPU ram crashes cockpit. On the mac this a major issue as the ram is shared and it just keep going.

4) I do most of my testing with a mac, and cockpit seems to work fine, except a few dark mode bugs.

5) Networking: I would go for one 10GB path per camera, but they could be on the same machine. Our diamond system has 2 cameras but with separate ethernet addresses and binding to different cards. This keeps things simple but may lead to confusion as people might not understand that the two parallel network links are doing different things. 

6) Windows speed: I have no idea why we found that strange speed issue for image transfer and saving on windows and maybe we should investigate it more seriously.

iandobbie, 2022-05-13:
The above info should probably get added to the documentation in some way. I will investigate.

iandobbie, 2022-05-16:
added into the main doc tree in 3066a842dd02a6f0687589a5a7c9b710c3b6f3a1

closed 2022-05-16 by iandobbie