microscope-cockpit/cockpit #785: Inverting z stage option

juliomateoslangerak, 2021-11-16:
Currently the Z stage is moving in a direct relation between Z position-analogue signal.
In an inverted system this is counterintuitive for the user. That is moving UP in the interface moves the stage UP and the focus is lower in the cells. While we might be used to this, for the users it is not intuitive as they imagine the cells with the coverslip in the bottom and it is difficult to imagine how things move in the machine.
Moreover, the Experiment interface shows the option 'current is bottom' as the relative position in the stack but not the 'current is top'. When marking multi-site experiments, most of the time the easiest and more reliable is to mark the coverslip side of things, and this is not possible in the current situation.
I looked into the piezo z-stage settings and could not find an obvious way to invert the axis.
I could not find this option in Cockpit/microscope either. 
Is it there? Would it make sense to implement it? If yes, who would be responsible for it? The PositionHandler?

I could look into implementing this.

iandobbie, 2021-11-16:
The axis mapper should allow you to trivially invert any axis. That said there are a million custom code lines as the x axis is inverted across all the defined areas like the mosaic, macrostage etc... so it might not be that simple!

iandobbie, 2021-11-16:
I just had a look and the nanomover and the PI687 device code has the axisSignMapper which would allow this but other devices don't seem to have this. It should be relatively easy to add it to the generic microscope stage axis device to allow the same thing to be used for generic microscope stage devices.

juliomateoslangerak, 2022-01-12:
So for the piezo, just modifying offset and gain does the trick. In my case, with a 200 um piezo from:
```
offset: 0
gain: 142.15
min: 0
range: 220
```
to:
```
offset: -220
gain: -142.15
min: 0
range: 220
```
Does it still make sense to implement this as a generic option?
Probably most controllers will allow an internal configuration option to invert axis...

iandobbie, 2022-03-15:
I suggest we close this issue, and revisit it if required in future. Inverting an axis "should just work" but might have a bunch of strange unexpected consequences.

juliomateoslangerak, 2022-03-15:
I agree the issue can be closed. I’ve been testing the solution for some time and had no issues. 
Should we add an example of this case in the doc string?