microscope-cockpit/cockpit #787: Mosaic view breaks when the camera is switched off

thomasmfish, 2022-01-10:
At Diamond we are testing the new version of cockpit on a new machine and we've found that the mosaic view completely clears and stops responding if the camera is toggled. I've attached images of the logging panels. The first screenshot is the error upon first toggling the camera, the second is the repeated error we press "Run mosaic" after the initial fault.
[image]
[image]
.

thomasmfish, 2022-01-10:
I should add I've tried using PyOpenGL 3.1.3b2 (which we have working with our previous version of Cockpit) and 3.1.4 and both have had this issue. PyOpenGL 3.1.5 would not load the mosaic view correctly (there was no location marker), so we weren't able to get as far as this error.

iandobbie, 2022-01-10:
Obviously this needs some testing but the 1286 error is what we find when the openGL canvas tries to create texture which is too large for the graphics card, also if you create too many textures (ie several thousand on the deepsim, or my mac laptop). What graphics card do you have?

The lack of a location marker is likely similar to several other issues where the mosaic window isn't properly updated. What happens if you move the stage? This often triggers the required redraw. I know this isn't a fix but shows what the fundamental problem is. There are a few places where this is an issue and it is somewhere on my todo list but not an actual issue yet.

iandobbie, 2022-01-10:
Just for a data point, on my mac in simulation mode (cockpit --no-config-files) disabling the camera makes no difference. If there is no camera enabled the mosaic puts up a modal dialog saying it needs a camera. 

There is something strange happening. Turning on and off a camera sometimes just works, sometimes crashes the mosaic entirely and sometimes just losses some of the tiles. If you zoom out to the required level to get the megatiles the images return. Obviously a bug that is sometimes loosing info but not always.

thomasmfish, 2022-01-10:
Error 1286 is only when trying to take images after openGL has already broken, even if only ~4 tiles were taken before it broke. It was breaking every time we tried switching the camera off after taking a few mosaic tiles. The new PC has a Nvidia Quadro P2200 5GB graphics card.

I'll need to test again tomorrow to see about moving the stage to update it but Chidinma was clicking on the mosaic view to get it to move and it wasn't responding, though I guess that's not surprising if it wasn't loaded properly.

thomasmfish, 2022-01-11:
From the meeting, it seems to be an issue relating to the display rescaling the macro tiles after the camera is cleared, so no longer has a min/max.

thomasmfish, 2022-01-14:
After all of that, it was freetype. Commit acece2390a45b343869fdf7b2b732d504beaf195 was the issue. Simply replacing `glDeleteTextures([self._texture_id])` with `del self` fixed the issue. I'll put a PR in to get a discussion going as to the best way to release the Glyphs

thomasmfish, 2022-01-14:
Hmm, no my bodge was bad (shocking). I don't know enough about OpenGL to figure this out now but it seems like there should be something fancier going on, [like this](http://bazaar.launchpad.net/~mcfletch/pyopengl-demo/trunk/view/head:/PyOpenGL-Demo/NeHe/lesson43/glFreeType.py).

iandobbie, 2022-01-16:
> 

> A fun new twist: it doesn't actually remove the tiles! The tiles can still be viewed if zoomed in, but if the mosaic view is zoomed out beyond a certain amount, it disappears. 

This is actually not surprising. There are in fact two sets of tiles, the full resolution ones and a zoomed out version refered to a megaTiles. We are obviously only corrupting one set, however I think it depends on what is being displayed when the camera is disabled as I found the high res tiles being lost rather than mega tiles.

thomasmfish, 2022-01-16:
Hi Ian, for me I'm pretty sure it was always the megatiles but I haven't tested that properly. Either way, I've put in a PR that seems to fix the issue (I think it was an OpenGL crash due to some textures trying to be deleted twice within too small a time frame (due to them being deleted by different threads).

This e-mail and any attachments may contain confidential, copyright and or privileged material, and are for the use of the intended addressee only. If you are not the intended addressee or an authorised recipient of the addressee please notify us of receipt by returning the e-mail and do not use, copy, retain, distribute or disclose the information in or attached to the e-mail.
Any opinions expressed within this e-mail are those of the individual and not necessarily of Diamond Light Source Ltd. 
Diamond Light Source Ltd. cannot guarantee that this e-mail or any attachments are free from viruses and we cannot accept liability for any damage which you may sustain as a result of software viruses which may be transmitted in or with the message.
Diamond Light Source Limited (company no. 4375679). Registered in England and Wales with its registered office at Diamond House, Harwell Science and Innovation Campus, Didcot, Oxfordshire, OX11 0DE, United Kingdom

closed 2022-02-03 by iandobbie

thomasmfish, 2022-02-08:
@iandobbie I've added a new commit to the branch you pulled that may be more to your liking, though I can't test it on the SIM and could never get the problem to replicate in the simulated system.

I don't understand what is wrong with using linked PRs to suggest changes for issues but I won't do that this time then

PR #788 (2022-02-03): Fix texture issues
This is the call that is causing the issues with the mosaic, though I'm not clear about why.

From what I can tell, there's no need to use `glDeleteTextures` on window close as textures that are exclusive to a context are automatically deleted with the context anyway. I'm assuming the frequent (but inconsistent) mosaic issues are due to timing often causing duplicate calls to `glDeleteTextures`, causing some break in OpenGL. As the mosaic would be the next obvious OpenGL update, that's where I noticed an issue. That could also explain why it often worked better the first tim [... cut]