microscope-cockpit/cockpit #795: Ambient Z-stack failing

thomasmfish, 2022-01-24:
It failed with the following error:
```

2022-01-24 15:20:12,948 INFO     experiment: 336  Experiment.execute started.
2022-01-24 15:20:13,496 ERROR    loggingWindow: 117  Exception in thread Experiment-execute:

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117  Traceback (most recent call last):

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117    File "C:\envs\cockpit\lib\threading.py", line 932, in _bootstrap_inner

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117      self.run()

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117    File "C:\envs\cockpit\lib\threading.py", line 870, in run

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117      self._target(*self._args, **self._kwargs)

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117    File "c:\b24\cockpit\cockpit\experiment\experiment.py", line 383, in execute

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117      raise RuntimeError("Found a line that no executor could handle: %s" % str(self.table.actions[curIndex]))

2022-01-24 15:20:13,496 ERROR    loggingWindow: 117  RuntimeError: Found a line that no executor could handle: (Decimal('0'), <light source named Ambient in group ambient>, True)
```
I have a fix coming

thomasmfish, 2022-01-25:
See #796 for a fix

carandraug, 2022-03-11:
This is a duplicate of issue #738. Closing as duplicated (still not fixed, let's discuss on the original issue).

closed 2022-03-11 by carandraug