microscope-cockpit/cockpit #803: Device specific depot config options.

iandobbie, 2022-03-15:
Revealed by my recent attempt to complete the multi-channel experiments on one camera and Julio's reversing Z stack orientation questions there is basically no documentation on any device specific config options in the depot config section. A few things that need to be addressed:

General description of params, URI, type, trigger source, trigger line,  analog line

Device specific config options:
stages: gains and offsets for analog, units per um for serial connections
cameras: filter wheels, multi channel experiment support
filter wheels: assignment of wavelength, labels, settle times

I'm sure there are plenty of others.

iandobbie, 2022-03-15:
I have added the above info plus quite a bit more. Need to merge the multi channel and the digital Z stack branches and add the relevant documentation as well. 

One issue is I have not done the spec for the ND filter wheel as I dont have an example and never used it in anger. We should check exactly what spec it is meant to have.

iandobbie, 2023-07-21:
The core current devices are documented. The point about needing to merge the digital Z stack and multi channel on one camera is still true. However these both have their own issues so I am closing this issue.

closed 2023-07-21 by iandobbie

carandraug, 2023-07-21:
For reference, they are currently documented at https://microscope-cockpit.org/depot-config.html