microscope-cockpit/cockpit #809: Stage homing at startup?

iandobbie, 2022-05-13:
This issue was prompted by writing code for the Ludl stage controller. This controller does not know the size of the stage until the stage has been moved it extremes.

My initial solution was the init the stage at microscope startup but not home it. Then home on connection to the device server, the problem with this was the limits were not properly defined on cockpit startup. I have hacked this by just homing the stage on microscope startup but I am not very happy with this solution. 

I think we need:

1) define if a stage needs homing - probably requires addition to the stage api
2) If a stage needs homing warn the users and allow them to relevant action to prevent crashes
3) perform home and then requiring the limits to ensure we know the true size of the stage. 

Not sure the best way to do this but we probably need  a stage specific extension to the microscope device server connection process.

carandraug, 2022-05-13:
If a `microscope.abc.Stage` needs homing, then that should happen when `enable` is called.  From the documentation:

```
    Some stages need to find a reference position, home, before being
    able to be moved.  If required, this happens automatically during
    :func:`enable`.
```

And [here's the line warning about it on Cockpit](https://github.com/MicronOxford/cockpit/blob/master/cockpit/devices/microscopeDevice.py#L564):

```
        # Enabling the stage might cause it to move to home.  If it
        # has been enabled before, it might do nothing.  We have no
        # way to know.
```

iandobbie, 2022-05-16:
Seems to me it would be beneficial to have a compulsory property to say if the stage needs homing. If the stage never homes then just set to false, If it needs homing then we can query this before executing the home command to allow notice to the user to shift the objective, sample or other safety measures. 

I'll try to implement this and see how it goes with my ludl test setup.

iandobbie, 2022-05-16:
I have suggested changes in cockpit:
https://github.com/iandobbie/cockpit/tree/home-stage 
and in microscope: 
https://github.com/iandobbie/microscope/tree/home-stage

This adds a single property to the microscope stage ABC, need_homed that by default just returns false and leaves the functionality as previously existed. Stages that need to be homed need to overwrite this property and also provide a home() function to do the homing (not needed if the stage doesn't require homing).

In cockpit if the need_homed returns true at startup then it puts up a modal dialog explaining that the stage will be homed and might move. If ok is pressed the stage is home, if it is cancelled then the stage is not homed, but weirdness may result.

These changes are trivial, yet provide support for stages that need homing, can be coded to properly support preserving home state across cockpit restarts, if microscope is left running and preserve the current behaviour if the stage doesn't need homing. 

I plan to pull these change into the respective master branches if no one objects in a few days.

iandobbie, 2022-06-08:
I have pulled these changes. Seems to work fine with no homing needed and on the ludl stage it displays the modal dialog at startup if needed.

closed 2022-06-08 by iandobbie