microscope-cockpit/cockpit #810: Why does every camera have readout mode and gain.

iandobbie, 2022-05-17:
The historic dominance of EMCCDs is reflected in the Cockpit camera GUI area. Every camera has a readout mode and gain control even through most cameras we use now don't have these options. I think these should all be folded into the setting dialog and used there. 

This would shrink the screen space needed and allow space to easily house a control to associate a a camera with a light source and help solve #681 and #216

carandraug, 2023-01-05:
Proposed fix on my `810-remove-readout-mode-and-gain` branch (it removes gain and readout mode from the UI but it's still available on the settings menu). Effectively, this commit https://github.com/carandraug/cockpit/commit/7691b1fadf7667d79f1177f9c86d500fba06d166

carandraug, 2023-01-10:
As discussed on today's meeting, I've pushed the proposed fix which removes the readout and gain from all cameras. We also discussed having select settings available on the main windows but I will open a new issue for that. Closing as fixed.

closed 2023-01-10 by carandraug