microscope-cockpit/cockpit #812: Iteration error when switching between cameras

thomasmfish, 2022-06-01:
This error has been showing up a lot here. It doesn't seem to have caused problems but it should be fixed. I've got a small update that I'll link to fix it.
[image]

thomasmfish, 2022-06-01:
I agree, I think a [queue](https://docs.python.org/3/library/queue.html) is probably the way to go, but I don't have time to sort that today and I only have today to deal with all the updates.

carandraug, 2022-06-01:
> I agree, I think a [queue](https://docs.python.org/3/library/queue.html) is probably the way to go [...]

I thought the same initially. However, grepping through cockpit code, it appears that `updateExposureTime` is only called through events, the same that calls `toggle` which is the only thing adding and removing cameras/lighs to those sets. So maybe some change how the events are called could prevent that race condition in a more natural way.

> [...] but I don't have time to sort that today and I only have today to deal with all the updates.

That is fair enough. But the end result of such approach is that issues get fixed with hacks piled on top of hacks which in the long run lead to something unmanageable.

iandobbie, 2022-07-14:
Looking at the code, the imager interface subscribes twice to the CAMERA_ENABLE event, with different functions. I think we probably just need to sequence the functions so that we only subscribe to the event once and this subscription fires both the toggle camera state and the updateExposureTimes (if required). I'll look at the logic and see if I can find a sensible refactoring.

iandobbie, 2022-07-14:
Both LIGHT_SOURCE_ENABLE and CAMERA_ENABLE  events cause calls to both the toggle function and the updateExposureTime function in the imager interface. It seems to me that we can just tag the call to updateExposureTime onto the end of the toggle function, so all the enable/disable is done and then the exposure times are updated which should solve the race condition.

carandraug, 2022-07-14:
Ian, your comments on the events made me remember that we already have an issue open for this. Closing this as duplicate of #603.

closed 2022-07-14 by carandraug