microscope-cockpit/cockpit #816: Need to be able to set hardware and software triggers to work from cockpit.

iandobbie, 2022-06-08:
We have discussed this point before but I think we do need a method in cockpit switch a generic triggered microscope device from software to hardware trigger mode.

For instance if you set a camera as hardware triggered in your depot.conf there is no way to then utilise a software trigger on this camera. Additionally by default the camera will probably start in software trigger mode as most devices seem to do that.

I think we need at least:

1) Method to switch a microscope device between hardware and software trigger modes that does so at both cockpit and microscope ends.

2) Method to set microscope device triggers to software or hardware at startup, presumable from their depot config. This should utilise the code in 1.

carandraug, 2022-06-08:
> This is really a cockpit issue so open as https://github.com/MicronOxford/cockpit/issues/816

I think you meant to close python-microscope/microscope#244 and keep this one open but did it the other way around. Reopening (and closing the issue in Python-Microscope).

iandobbie, 2023-10-05:
I have done some investigating on this for the picam (a microscopeCamera device) and the issue is that the relevant parameters are on set on device init. The code in the camera handler is:

```
        # Set up trigger handling.
        if trigHandler and trigLine:
            h = trigHandler.registerDigital(self, trigLine)
            self.triggerNow = h.triggerNow
        else:
            softTrigger = self.callbacks.get('softTrigger', None)
            self.triggerNow = lambda: softTrigger
            if softTrigger:
                depot.addHandler(cockpit.handlers.imager.ImagerHandler(
                    "%s imager" % name, "imager",
                    {'takeImage': softTrigger}))
```

We need to switch on the trigger mode not on the existance of the trigger line and device, although check for these. Then we can do the setup in the init code as is. The we need a new function to call when we change trigger mode which sets the triggerNow function. 

I dont think this will change the behavious in experiments if we have a digital line defined but will allow soft trigger outside experiments if needed. 

I will implement for microscopeCamera and check to see it works and then think about spreading to other digital triggered devices.