microscope-cockpit/cockpit #821: Remote camera not working

dakoner, 2022-08-18:
I am testing a new camera type in python-microscope and trying to use cockpit to view the results.
I had trouble with my camera type so I tried this depot file:
```
[remote]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:SimulatedCamera@192.168.1.150:8000

[local]
type: cockpit.devices.dummies.DummyCamera
```

The simulated camera is on another machine with this python-microscope config:
```
import microscope
from microscope.device_server import device
from microscope.cameras.opencv import CVCamera
from microscope.simulators import (
    SimulatedCamera,
)

DEVICES = [
    #device(CVCamera, "192.168.1.150", 8000,), 
    device(SimulatedCamera, "192.168.1.150", 8000),
]

The local DummyCamera works, but the simulated camera doesn't.  I can see that the remote simulated camera is getting invoked (I added debug logs showing it returns a numpy array), but the callback that should be triggered to invoke MicroscopeCamera's receiveData method never calls.  As a result, the camera view never updates with the simulated camera.  This is the same thing I see from my CVCamera, so I'm checking with the maintainers to see why this wouldn't work even for a simulated camera.

I'm running on Windows,  running from git master,

iandobbie, 2022-08-18:
I have a slightly different config as I only have one machine, so I am running microscope and cockpit on the same machine, and the network address is then 127.0.0.1. I have removed redundant lines to test so my microscope config becomes (in file sim-cam-test.py):

import microscope
from microscope.device_server import device
from microscope.simulators import (
    SimulatedCamera,
)

DEVICES = [
    #device(CVCamera, "192.168.1.150", 8000,), 
    device(SimulatedCamera, "127.0.0.1", 8000),
]

which is then run with 

deviceserver sim-cam-test.py

The cockpit config isn file sim-cam-test.depot is:

[remote]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:SimulatedCamera@127.0.0.1:8000

which is then run with 

cockpit --depot-file sim-cam-test.depot

This is all on a mac laptop and it works fine. 

I suggest you start with the above config and see if it works on a single machine. If this works move the microscope config to the other machine and edit the IP address in the two files and see if that works. One possible issue might be firewalls, on either machine as the camera very much needs two way connectivity with both side initiating new connections.

Just realised that if you don't have a server section in your cockpit depot file this will not work. 

You cockpit depot file needs a section like:

[server]
ipAddress: 10.1.10.186
port: 7700

This includes the ipaddress of the machine running cockpit, and is sent to the microscope client when the camera connects. I have just realised that this does not appear to be documented anywhere, Sorry!

iandobbie, 2022-08-18:
I strongly suspect this is the lack of the server section. I have added some documentation on this (see issue #882)

dakoner, 2022-08-18:
OK, I started by running a SimulatedCamera on localhost and then connecting cockpit to that.
That does work.  

Next, I tried CVCamera on localhost and connecting cockpit to that.
That fails, but with an exception that indicates it did in fact receive the image (1280x720, 3 channels) and errors because cockpit is expecting a 2-dimension, not 3-dimension.  Fixed that.

So the next question I need to resolve is whether adding the server section makes real remoting work.

iandobbie, 2022-08-18:
Glad the local host version works. If the remote option doesn't work with the server section added I would strongly suspect firewall rules of some sort as this has caused us problems many time before.

As to the openCV multi channel image there are two relatively simple solutions,

1) implement different colours as a filter wheel on the "camera", see the stage_aware_camera.py  module in the simulators directory. 

2) implement the 3 colours as 3 independent cameras, all triggered simultaneously.

Extending the camera module to understand 3  colours images natively would be more work than either of these. 

If you use method 1, there is code to allow experiments with multi-bandpass filter sets a rapidly triggered multi-line LED sources which would allow you to effectively just return 3 sequential images and have them end up as 3 separate colour channels in the output image, however this doesn't work in the "snap" image mode.

dakoner, 2022-08-18:
You were right!  I had to add the server section to the cockpit config (which causes Windows to throw up a "hey you're starting a server" window).  Then both the remote simulated and OpenCV cameras work.  Now it's time to implement my grbl stage in microscope-python.

BTW my own code (a microscope gui https://github.com/dakoner/microscope_ui/blob/main/microscope_ui.py) does this to support 3-channel:

        image = QtGui.QImage(draw_data, draw_data.shape[1], draw_data.shape[0], QtGui.QImage.Format_RGB888)

which works directly with images of the type returned by color mode in OpenCV.

closed 2022-08-18 by dakoner