microscope-cockpit/cockpit #829: New release needed?

iandobbie, 2022-09-27:
With the recent issue relating to cockpit and microscope version incompatibility I think we need to release new versions of both Cockpit and Microscope. Maybe we should have a plan to release new versions on  a set schedule. 

Any thoughts?

juliomateoslangerak, 2022-12-13:
I thought we might profit from this release to change a number of things
- Black our code #649
- We probaly should use pre-commit to configure the 'blacking' of code before any commit.
- Maybe use isort in a pre-commit too
- I'm currently using poetry as a package managment system in another project. It's very nice

iandobbie, 2024-03-21:
David produced a new release on pypi on 2024-01-10, along with a synchronized release of python-microscope. Closing as complete

closed 2024-03-21 by iandobbie