microscope-cockpit/cockpit #833: Small bits of old code that my IDE picked up

thomasmfish, 2022-11-10:
I've made a branch of some changes [here](https://github.com/thomasmfish/cockpit/tree/some-tidying) that are small and could easily be cherry-picked in if that'd be helpful. Nothing that has caused me issues but a few things there could break error reporting.

carandraug, 2022-11-10:
I've merged four of those but I don't think the other two are quite right. I'll leave a comment on the individual commits.

thomasmfish, 2022-11-10:
Great, thanks!

carandraug, 2023-03-23:
I'm not sure what happened here. I though I had left comments on the commits that I didn't cherry-pick but I can't find them anymore. So I'm adding them here:

* 94638a3941303e87a0b70e7e2a85dbcc84ab4867 and 3eab06425eca33feeb9adda22ca11f1f6e427390 (the second overwrites the first): I didn't pick these because while it technically fixes the code by not using the inexistent variable, it's not what the code was meant to do. I've looked through the history and left a longer explanation on the commit message of fix that I eventually did commit (which is bffc2aa87b25304da8c3af03ed7b8920dd15cd7c )
* 1c5ec9c1c35439b90152bd20d5876d061d18f87a : I didn't pick this because I don't think the change makes the code any better. It simply ensures that the file is closed in a `try/catch` but the file will already be closed since this at the very end of the function. The block could be improved for sure but I don't think this change was a improvement in the right direction.

The four commits that I picked were:

* c12606ad68fc6f47dc13422d48f27579c22bb1c2
* 2d4c9492bbce4106620470c458fc238c698629d6
* 6c507b82ccd17fe961b20b30178f23373e0d49cf
* d1c8bcd21e894fdbd04a374a45c65823566b75f4

Closing as fixed.

closed 2023-03-23 by carandraug

carandraug, 2023-03-23:
I've found the comments now (they're on the commit but only if you access through the URLs `https://github.com/MicronOxford/cockpit/commit/commit-hash` - if you try `https://github.com/[thomasmfish](https://github.com/thomasmfish)/cockpit/commit/same-commit-hash` they're not there).

Anyway, there was a small detail this comment https://github.com/MicronOxford/cockpit/commit/1c5ec9c1c35439b90152bd20d5876d061d18f87a#commitcomment-89525659 about replacing `file` with `open` which I missed. I've now also merged it.