microscope-cockpit/cockpit #836: Use of shifted average histogram needed?

juliomateoslangerak, 2022-12-13:
After running some CPU profiling tool it seems that this function is taking, by large, a good deal of the CPU. Presumably iterating over m to get a shifted average histogram is eating a lot of CPU.
Do we really need a shifted average histogram? This is only to display a histogram under the image, right?
```python
def setData(self, data):
    # Calculate histogram.
    # Use shifted average histogram to avoid binning artefacts.
    # generate set of m histograms
    #   each has class width h
    #   start points 0, h/m, 2h/m, ..., (m-1)h/m
    # 1 < m < 64
    # sum to average
    if self.lbound is None:
        self.lbound = data.min()
    if self.ubound is None:
        self.ubound = data.max()
    if self.lthresh is None:
        self.lthresh = self.lbound
    if self.uthresh is None:
        self.uthresh = self.ubound
    nbins = 64
    m = 4
    self.bins = np.linspace(data.min(), data.max(), nbins)
    self.counts = np.zeros(nbins)
    h = self.bins[1] - self.bins[0]
    for i in range(m):
        these = np.bincount(np.digitize(data.flat, self.bins + i*h/m, right=True), minlength=nbins)
        self.counts += these[0:nbins]
```

carandraug, 2022-12-14:
> [...] This is only to display a histogram under the image, right?

I believe so, yes. Does replacing it with something like `np.histogram` makes things any faster for you?

iandobbie, 2022-12-14:
I suspect this is Mick's work. The histogram used to have nasty aliasing meaning at time 1 in n peaks was 1/n higher depending how the data was divided between the histogram bins. This was very ugly and the autoscaling of the histogram meant it continually changed. The newer histograms are much cleaner, but maybe we need a lazy histogram where it is only calculated and redisplayed if the image doesn't change for some period.

juliomateoslangerak, 2026-11-09:
I pushed this commit 18dfd05 to the refactored branch where I tried this other algorithm to compute the histogram.

Instead of computing multiple times the histogram with different shifts, it does compute the histogram once using a finer binning and applies a moving median kernel to the histogram. I also preallocate some of the variables in the class so that they are not generated each call.

I tested performance and, on a 1024x1024 image, it is somewhere between 25x and 30x faster (140 msec vs 5 msec). Given the amount of calls to this function I think this might be a major improvement in performance. The histogram looks fine to me, but I guess it has to be tested in other scenarios that simulated images.