microscope-cockpit/cockpit #841: Permissions are not kept when reordering SIM

thomasmfish, 2023-01-19:
This has caused access issues. I've proposed a change in this branch:

https://github.com/thomasmfish/cockpit/tree/copy-permissions-when-reordering-SIM

iandobbie, 2023-01-24:
This seems pretty trivial so I pulled it and pushed to master. Updated the commit to reference this issue.

closed 2023-01-24 by iandobbie

thomasmfish, 2023-02-13:
@iandobbie Thanks! I may have a problem though - it has been working but not reliably.

I assume this is a Windows only issue with `shutil.copymode` because the permissions on Windows are weird, but occasionally it's ended up with unreadable permissions that need admin rights to fix. As a result, I've updated the branch with a potential fix that does everything a little slower and uses `shutil.copyfile`, then removes the temp file, as (unlike the other copy/move functions) `shutil.copyfile` doesn't copy permissions so permissions should be inherited from the target directory (unfortunately, defining `copy_function` in `shutil.move` doesn't do it). I'll try loading this onto the SIM when I get a chance and see if that helps.

iandobbie, 2023-02-13:
This is very weird, but I can imaging that a race condition could affect the final permissions. Seems like a nasty OS bug though and likely to produce horrors in lots of vital OS functions. Please do update us if you get a reliable fix and we will commit it.