microscope-cockpit/cockpit #843: Mac macrostage data display for position and steps size are too small.

iandobbie, 2023-01-26:
The mac macrostage window looks like the image below, chopping of the actaul stage position and step size.
[image]

iandobbie, 2023-01-27:
I hacked this by making the text boxes a fixed width but this isn't a good solution as it bypasses the sizer layout engine.  How do we make the sizer give them the right size to start with and why are the saved top and saved bottom positions wider?

thomasmfish, 2023-02-28:
Looking at the code (I'm not able to test this), I have a few suggestions that may help:

- Call `sizer.Layout()` just before `self.SetSizerAndFit(sizer)`
- Change `sizer.Add(stage_sizer)` to `sizer.Add(stage_sizer, 1)`
- Try adding proportion of 1 to various things when adding them to sizers, including when adding sizers to sizers. Doing this for everything will likely go wrong, as (I think) it'd make the buttons equal proportion to the stage view, but setting proportion to not 0 changes some behaviour beyond just setting a relative size (hence the previous suggestion).

iandobbie, 2023-03-02:
None of this worked because the text boxes where empty when the size was called. As the text format is fixed as 5.2f and 4.2f for the position and step size respectively I prefilled them with 00000.00 and 0000.00 at creation and it all just worked. Fix is in https://github.com/iandobbie/cockpit/tree/macrostage-text-fix.
The actual values are pulled before you notice anything, so I think this is a pretty good fix. I will test on a PC and then push into the main branch. 

 I will test it on the PC but I'm pretty sure this will just work

iandobbie, 2023-03-02:
Good suggestion, Not quite got it to work yet, I'll have a fiddle and see if I can make it work.

iandobbie, 2023-03-02:
What I settled on is 
        self.SetInitialSize(self.GetSizeFromText("88888.88"))
in the position control and with one less 8 in the step size control.
Seems to work on the mac, will test on PC and push to main branch

thomasmfish, 2023-03-02:
Can you not set it like this?
`super().__init__(parent,size=self.GetSizeFromText("88888.88"),style=wx.TE_RIGHT|wx.TE_READONLY)`
The init for `wx.TextCtrl` is `__init__ (self, parent, id=ID_ANY, value=””, pos=DefaultPosition, size=DefaultSize, style=0, validator=DefaultValidator, name=TextCtrlNameStr)`, so size should be settable there

iandobbie, 2023-03-02:
I did try that first but it didn't seem to work. I'll try again.

iandobbie, 2023-03-02:
Nope definitely doesnt work for me....

File "/Users/ID/src/cockpit/cockpit/gui/macroStage/macroStageWindow.py", line 67, in __init__
    super().__init__(parent,size=self.GetSizeFromText("88888.88"),style=wx.TE_RIGHT|wx.TE_READONLY)
RuntimeError: super-class __init__() of type HandlerPositionCtrl was never called

But no other obvious error messages.

thomasmfish, 2023-03-02:
Oh, I think `GetSizeFromText` probably depends on the style, so can't be called until after `__init__` - sorry about that! In that case would `self.SetMinSize` be best? I'm not sure what initial sets but I'm guessing a minimum size is the most important size?

iandobbie, 2023-03-02:
Not sure but boxes obviously aren't growing to the size of the text on macos, so I think setting initial size is ok, I'll what set min size does.

iandobbie, 2023-03-02:
So SetMinSize produces exactly the same result as the SetInitialSize so I guess it is preferential as of something later wants to change the size there will still be a sensible minimum.

thomasmfish, 2023-03-02:
Yeah, my guess is you could end up with squished boxes, depending on some flags and resizing, with `SetInitialSize` but they probably both practically do the same if `wx.EXTEND` is not set as a flag.

iandobbie, 2023-03-02:
Ok, I just checked it on windows and it works fine there so I'll push it. Closing this discussion

closed 2023-03-02 by iandobbie