microscope-cockpit/cockpit #847: wavelength colors not showing up in main window

juliomateoslangerak, 2023-02-15:
I've been messing around with merges and changes in the depot files and I cannot get the wavelength colors to show up in the main window. I presume there is something wrong with my depot. The colors for the cameras are showing up correctly in the cameraview and for the lasers in the touchscreen. I presume it might have to do something with the ND filter wheel... or mayb ethe drawer... here is a fragemnt of my depot with the relevant devices
Are ```wavelength``` and or ```dye``` relevant as a camera properties?

```ini
...
[Drawer]
type: cockpit.devices.drawer.Drawer
cameras: Green-Cam, Red-Cam, Finder-Cam
default: Green-dye:525, Red-dye:600, Finder-Cam:0
far-red: Green-dye:525, Red-dye:646, Finder-Cam:0

[Green-Cam]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:AndorSDK3@10.6.19.31:8001
# transform: (lr, ud, rot)
transform: (1, 1, 1)
# ROI: (left, top, width, height)
ROI: (513, 513, 1024, 1024)
triggerSource: cRIO
triggerLine: 4
dye: eGFP
#wavelength: 525
settings:
   pixel_readout_rate: 100 MHz
   simple_pre_amp_gain_control: 16-bit (low noise & high well capacity)
   trigger_mode: External Exposure

[Red-Cam]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:AndorSDK3@10.6.19.30:8001
# transform: (lr, ud, rot)
transform: (0, 0, 1)
# ROI: (left, top, width, height)
ROI: (513, 513, 1024, 1024)
triggerSource: cRIO
triggerLine: 0
dye: RFP
#wavelength: 646
settings:
   pixel_readout_rate: 100 MHz
   simple_pre_amp_gain_control: 16-bit (low noise & high well capacity)
   trigger_mode: External Exposure

[Finder-Cam]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:XimeaCamera@10.6.19.30:8010
dye: None
#wavelength: 0
# transform: (lr, ud, rot)
transform: (0, 0, 0)
# ROI: (left, top, width, height)
# ROI: (513, 513, 1024, 1024)
exposure_time: 0.12
settings:
   gpo_mode: XI_GPO_EXPOSURE_ACTIVE
   imgdataformat: XI_MONO16
...
[Transmission]
type: cockpit.devices.light.SimpleLight
triggerSource: cRIO
triggerLine: 3

[488nm]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
uri: PYRO:DeepstarLaser@10.6.19.21:9011
triggerSource: cRIO
triggerLine: 5
wavelength: 488

[nd-filters]
type: cockpit.devices.microscopeDevice.MicroscopeFilter
uri: PYRO:ThorlabsELLSlider@10.6.19.21:9014
lights: 488nm  #, 561nm, 647nm
filters:
  0: ND 100%%
  1: ND 32%%
  2: ND 10%%
  3: ND 1%%
...
```

iandobbie, 2023-02-15:
[image]
In the attached image you see the blue bar between the camera and the settings button in the Cameras section. This should come from the camera wavelength ( and should match the colour of the camera title bar in Camera Views menu. I will see if I can simulate enough to copy your setup and see if I can reproduce the issue here. 

Another contribution might be the drawer code. Although this has been preserved we haven't used this functionality since the version running on the OMX v2 in Oxford.

iandobbie, 2023-03-01:
I just setup a test system with a couple of simulated cameras and the following config file and 2 simulated cameras on localhost copy of microscope. 

[server]
ipAddress: 127.0.0.1
port: 7700

[cam1]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:TestCamera@127.0.0.1:8000
transform: (1, 1, 1)

[cam2]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:TestCamera@127.0.0.1:8001
transform: (1, 1, 1)

[Drawer]
type: cockpit.devices.drawer.Drawer
cameras: cam1,cam2
default: cam1:525, cam2:600
far-red: cam1:525, cam2:488

This works fine on my Mac. I will try on the PC.

iandobbie, 2023-03-01:
I didn't notice before but now I see that the light sources also don't have the correct colour bars. See image from my mac which has the camera colour bars but also the light sources all have colours as well. 

[image]

iandobbie, 2023-03-01:
Quick update seem that the wx.Control.SetBackgroundColour just doent do anything on windows. I wonder if we can fake it? Or if we have to write our own widget just to get this coloured line!

iandobbie, 2023-03-01:
On the PC if you change from wx.Control to wx.StaticText(label="") it works.

iandobbie, 2023-03-01:
This is fixed on mac and PC by the above change in my branch colour-bar-fix. Can someone test on a linux system?

iandobbie, 2023-03-08:
Closing as fixed

closed 2023-03-08 by iandobbie