microscope-cockpit/cockpit #849: Weird issue with logviewer.

iandobbie, 2023-02-20:
I'm on a mac with Python 3.10.9 and trying to access the logviewer window gives me

Traceback (most recent call last):
  File "/Users/ID/src/cockpit/cockpit/util/csv_plotter.py", line 27, in <module>
    import matplotlib
ImportError: No module named matplotlib

However from the command line or from the pyshell window within cockpit an "import matplotlib" works fine. Any suggested debug approaches?

carandraug, 2023-02-20:
Does `import cockpit.util.csv_plotter` work from the pyshell window? If you add `print(sys.path)` at the top of `cockpit/util/csv_plotter.py` (before the `import matplotlib`) does it differ from the `sys.path` you get when you run it on the pyshell?

iandobbie, 2023-02-20:
The weirdness continues. The import works find from the pyshell, but also the print(sys.path) in csv_logger includes ‘/opt/homebrew/lib/python3.10/site-packages' which is where the matplotlib is located.

All very weird. I'm sure its a trivial issue but I don't understand what the issue is.

iandobbie, 2023-02-20:
Ok, getting nearer the bottom of this.... On startup csv_plotter is obviously evaluated and print a sys path for python 3.10, however when I actually try to instantiate the class by starting the logViewer it start in system default python2.7! Which obviously doesn't have matplotlib installed. The path becomes

['/Users/ID/src/cockpit/cockpit/util', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python27.zip', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages']
Traceback (most recent call last):
  File "/Users/ID/src/cockpit/cockpit/util/csv_plotter.py", line 29, in <module>
    import matplotlib
ImportError: No module named matplotlib

iandobbie, 2023-02-20:
The code in question is 

        if not logs:
            menu_item.Enable(False)
        else:
            shell = sys.platform == 'win32'
            args = ['python', csv_plotter.__file__] + logs
            self.Bind(wx.EVT_MENU,
                      lambda e: subprocess.Popen(args, shell=shell),
                      menu_item)

So it checks if there are any logs to display, then starts a shell calls "python" which will start the system python on my mac. Not sure what the best fix is.

I have bodged my system to just call "python3" which shows other version rot errors in the valueLogViewer.

carandraug, 2023-02-20:
Nicely found. I've pushed a fix that uses `sys.executable`. Closing as fixed.

closed 2023-02-20 by carandraug