microscope-cockpit/cockpit #850: DummySIM isn't available

thomasmfish, 2023-02-22:
For some reason the dummy SIM that is in Microscope isn't in Cockpit. I've made the change in a branch on my fork (see https://github.com/MicronOxford/cockpit/commit/fa509b4003cf9523b40dd01d8ee0945e89ad32b3), though I was only after it for GUI testing purposes

iandobbie, 2023-02-24:
You are correct we never implemented dummy SIM in cockpit as basically it requires the hardware triggers to synchronize all the hardware and it just seemed too much effort for little benefit.

thomasmfish, 2023-02-27:
Ah okay, I was only after another GUI elements to test how resizing & minimum sizing worked with wrapping the elements

thomasmfish, 2023-02-27:
Feel free to close this then!

iandobbie, 2023-02-27:
Seems pretty trivial though so introducing it is pretty easy. 

There is now also a digital IO device in microscope and cockpit so we could easily add a dummy DIO device