microscope-cockpit/cockpit #852: Main panel not sizing with window properly

thomasmfish, 2023-02-22:
I'm not sure whether some of the components need a minimum size set or there is a fancier way to update the minimum dimensions of the Window, but I've been trying some stuff on the branch here:
https://github.com/thomasmfish/cockpit/tree/ensure-GUI-refresh

I started looking into this because a user was having issues with certain GUI elements being cut off and not updating properly but I think it could use more work than what I did for the first commit 6ab5d319b1d882514c996fd6c071e9739fe66603

iandobbie, 2023-02-24:
I was having weird issues with the text boxes in the stage panel on a mac (see issue #843) which don't happen on the windows version. GUI sizing seems to be a bit of a dark art, and probably needs to be test across implementations are different bits are implemented differently across Mac, Windows and Linux.

thomasmfish, 2023-02-28:
I've done a bit more work on this and I noticed there were a few places where a sizer wasn't set, causing issues. I'm 90% happy with what I've got in that branch now, though for some reason the `SetPointGauge` for light power gets a little squashed (perhaps due to wx ignoring the size of the status bar?). That branch is built on the changes I mentioned in #851, though I'm not sure there's any actual requirement for that (I just didn't want the lack of it to be the root of any issues).

thomasmfish, 2023-03-14:
[image]
Thought I'd add an image showing what happens on Windows

thomasmfish, 2023-03-14:
[image]
And with https://github.com/thomasmfish/cockpit/tree/ensure-GUI-refresh

iandobbie, 2023-03-14:
On Mac you don't get the nasty borders but none of the controls move on resize

iandobbie, 2023-03-15:
Just tried this on the Mac and it appears fine. Hopefully David will check it out and give it a quick test on Linux then we can merge it into the main branch.

thomasmfish, 2023-03-17:
Great, thanks!

thomasmfish, 2023-03-29:
[image]
There does appear to be a problem with the filters that's popped up from this

thomasmfish, 2023-03-29:
Also if I really muck with resizing this happens to the SLM:
[image]

iandobbie, 2023-03-29:
The joys of GUI development. The filters probably need a different min size or something. The SLM is weird. Do the lights go the same way if you make them huge? Also what types of controls are the SLM buttons? Are they the same as the camera and lights buttons?