microscope-cockpit/cockpit #863: Unassigned variables in PicoMotorDevice

thomasmfish, 2023-04-24:
Unassigned in `homeMotors`:
`endpositon`
`oldposition`

Not an issue for me, just thought I should report it

iandobbie, 2023-04-26:
As far as I know the only system actually using this device was the original Oxford CryoSTORM system. This was dismantled some time ago and I certainly don't have access to the hardware so I'm a little reluctant to fix them.

Anyway the device should be moved into microscope, I wonder if this is the moment to delete it from cockpit maybe leaving it in a legacy device branch so we don't accidentally loose the code if we need it again in future?

iandobbie, 2023-04-26:
Just looked and the fix is pretty obvious. The oldposition should be origPosition and the assignment for endposition had a typo in the name. No method to test this so just checking it in. 
Commit  d92b79b2

closed 2023-04-26 by iandobbie