microscope-cockpit/cockpit #870: does joystick segfaults in Mac

carandraug, 2023-08-17:
I was looking at `cockpit.gui.joystick` and found this:

```
class Joystick:
    def __init__(self, window):
        if sys.platform == 'darwin':
            return None
```

which I found weird. This was added in b3633edbe63a which just moved the check from `MainWindow` to `Joystick`. The check against Mac was added in b0ac81db0de4 which says it was done because it was segfaulting in Mac but that was back in 2018 and maybe the issue has been fixed since.

Also, I remember Mick testing some inconsistencies on joystick between mac, linux, and windows so at least on his tests it worked on Macs. Maybe the issue that @iandobbie had in 2018 was that his build of wxWidgets was not built with joystick support? If so, skipping Mac's is wrong and instead we should check if support for joystick exists. Looking at wxPython's demo for joystick it seems that checking `wx.adv.USE_JOYSTICK` would do that.

@iandobbie , can you check if `j = wx.adv.Joystick()` works in your system and what is the value of `wx.adv.USE_JOYSTICK` ?

iandobbie, 2023-08-17:
In the pyshell I did
import wx
from wx import adv
j=adv.Joystick()
j
<wx._adv.Joystick object at 0x14fe4f880>

So it at least initialised. 

I thought the joystick code had been removed to be honest. I will see if I can get it to work on my mac with a console controller.

iandobbie, 2023-08-17:
With no joystick connected if I comment out the if darwin return clause then cockpit startup crashes somewhere in the wx joystick code. 

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libwx_osx_cocoau_core-3.2.0.0.0.dylib	       0x1055e1dec wxJoystick::GetXMin() const + 4
1   _adv.cpython-311-darwin.so    	       0x107e8e140 meth_wxJoystick_GetXMin(_object*, _object*) + 108
2   Python                        	       0x104e2068c cfunction_call + 96

So the call to find XMin crashes. 

Ian

iandobbie, 2023-08-17:
Just tested if I create a stick then call stick.XMin it crashes. So need to know if we actually have a joystick connected at least for darwin (macos).

carandraug, 2023-08-17:
That's good enough for me. It shows that either at least on Mac it no longer fails. Either it has since been fixed upstream or you just had a wx build that was built without support for it. I've replaced the check against mac with a check against support for joysticks in wx.

Happy to also have the whole thing removed if it's not useful.

carandraug, 2023-08-17:
ups. I was wrong.

carandraug, 2023-08-17:
I'm guessing that the error was happening because you don't actually have a joystick connected and we can query that from wx (which is also what the wx demos do). So I pushed 64bba9a4 to do that.

closed 2023-08-17 by carandraug