microscope-cockpit/cockpit #873: Joystick sensitivity controls?

iandobbie, 2023-08-26:
I paired my Mac with an xbox conttoller and a bit of trial and error got it to work. Tap of "A" does a find stage in the mosaic. The left joystick moves the mosaic view around and holding A while using the left joystick moves the stage.

However the control is way too sensitive, you can go from one side of the stage to the other in less than a second. I think we need some sensitivity control, or at least a config parameter that can change it so it is usable.

iandobbie, 2023-09-26:
Added a config parameter to set this system wide and wrote some documentation. Added in a few commits:

9e75616e252cc89ef1334294b12e694751c16880
96ed3833f982b73878f9258faf3331bd629474ee
dd8402bacb5213d4b8caae3c2a6df58af1ad8a72

closed 2023-09-26 by iandobbie

carandraug, 2023-09-26:
Please reference the issue number on the commit message since this enables finding discussion (if any) on the bug tracker easily. If in the future someone needs to find where a discussion for some part of the code is, then they find the commit that introduced the change and then have the commit message referencing the bug tracker.

carandraug, 2023-09-26:
Anyway, I think default configuration values are supposed to go in ` cockpit.config._default_cockpit_config`

carandraug, 2023-09-26:
> Anyway, I think default configuration values are supposed to go in  cockpit.config._default_cockpit_config

Now done.