microscope-cockpit/cockpit #877: Integrating new devices

joris-afk, 2023-09-20:
Hi,

I'm a McGill student working under Mike Strauss as part of my Capstone project for my undergraduate studies. I'm looking to expand the devices compatible with Cockpit to work with a pco.panda camera and an ASI ms2000 xy stage we have. What files do I need to add for each device? In which directory would they need to go in order to respect the program's architecture? Will the microscope and/or cockpit python packages also need to be modified or can I just write my files in the microscope-cockpit folder directly after forking this repository and downloading it to my local device? 

Thank you for your time,
Joris

iandobbie, 2023-09-20:
Hi Joris, 

Cockpit talks to hardware via a unified API based around device type. The  hardware support is actually provided by python-microscope (https://github.com/python-microscope/microscope). Once these devices are supported in microscope you will just need to write a config file for microscope that says what hardware is where, and then in cockpit to say how to connect to each device and then any optional parameters, stage calibrations, camera orientations etc...

As a first test I would just download and install cockpit and then try it out with simulated devices by just running with no arguments which will start a system with 4 simulated cameras, xyz stage and a few light sources.

iandobbie, 2023-09-22:
Just to say that I merged the ASI stage code into microscope (https://github.com/python-microscope/microscope/issues/283) for the latest status of the support.

joris-afk, 2023-09-23:
Alright. Thank you for keeping us informed.

joris-afk, 2023-10-18:
Just keeping you updated on our progress, we are currently in the testing phase for the stage. We managed to get it move through cockpit. We're only missing a few more test to be sure it wasn't a fluke and maybe adapt the code to consider a z-axis.

iandobbie, 2023-10-18:
Sorry my system doesn't have a Z axis but you should just be able to add it as an axis in cockpit, but I think the microscope code has a hard coded limit, see https://github.com/python-microscope/microscope/issues/287