microscope-cockpit/cockpit #881: Alternative file formats

iandobbie, 2023-09-30:
Cockoit only currently deals with .dv files, pretty obscure. We should consider alternative file formats, the most promising I think is the ngff which has a python library implementation and will store the relevant metadata. 

This should be linked up with the expansion of metadata in non-experiment image files, like snaps or mosaic images. This is an ongoing effort see issues #676 and #391.

juliomateoslangerak, 2023-10-03:
A recent post of Josh Moore in image.sc on the status of NGFF
https://forum.image.sc/t/ngff-updates-oct-2023/86957

iandobbie, 2023-10-12:
So I have implemented a function to save snaps from the camera views as a zarr file. So far contains no metadata as I couldn't work out how we were meant to add it. However it does save the data into a zarr file which can then be loaded in napari via the open folder menu item. 

Code is in my zarr-snap-save branch at the moment. I will research the metadata to see how to get that stored with the file.

juliomateoslangerak, 2023-10-13:
The OME metadata is stored as a ome.xml file in the group level
https://ngff.openmicroscopy.org/latest/#bf2raw-details
There is a library to easily produce these ome xml files that might be worth considering… I’ll look for it.

iandobbie, 2023-10-18:
I asked on image.sc (thread at https://forum.image.sc/t/microscopy-metadata-in-zarr-files/87399) and was pointed at:

https://github.com/ome/ngff/issues/104 and https://github.com/tlambert03/ome-types

I plan to explore the python libraray and try to get as much metadata as we can into the ome-xml format for storage into the zarr.

iandobbie, 2024-10-22:
Definitely looks possible. A quick trawl of the README etc found that the zarr writer might well be in development rather than a finished project. I will take a look when I have some time.

juliomateoslangerak, 2025-04-26:
I'm looking into implementing ome-zarr saving. I looked into a number of libraries providing zarr image write and read:

- BioIO: (https://github.com/bioio-devs/bioio. formerly AICSImageIO AllenCellModeling/aicsimageio): It looks nice and a lot of functionality but it brings quite a number of heavy dependencies.
- ngff-zarr: Interesting too (more parallelisation) but, again also too many dependencies.
- ome-zarr-py: (https://github.com/ome/ome-zarr-py) Default ome-zarr implementation layer. Easy and it garanties ngff compatibility but not quite enough. It does not provide the append method (stream new images to an existing dataset). and also some extra dependencies like dask.
- tensorstore: Google developed library in C (real parallelisation) with a python api. Low level and build for performance.
- python zarr: Zarr-Python the standard implementation of zarr in python. This one is low level and base for all the other tools. Provides all the necessary tools and minimal dependencies. 

I talked to Josh Moore and he recommended ome-zarr or tensorstore for performance. 

We can use ome-zarr to generate the initial dataset and then zarr to append data. That said, given the limited use of cockpit, it might be just better skip ome-zarr and go directly to vanilla zarr.

If performance becomes an issue, we can go for tensorstore. I did some preliminary tests and, I believe that vanilla zarr is enough for most use cases. Again there is some tweaking to get the performance right. My initial (not very consistent) tests confirm what is expected:
- compressing (Blosc) is faster than not compressing
- chunking adapted to image sizes
- Preallocating and then replacing data is somehow making things very slow
- I didn't test that but I think it might be best to construct a full time point (repetition) array in memory and then save that at once as a whole chunk or (now possible) chard.

There is this interesting [thread](https://forum.image.sc/t/ome-zarr-chunking-questions/66794) in image.sc too, but a bit old by now

iandobbie, 2025-04-26:
I also started to look at this and hit the major hurdle about how to store the metadata. This is still not defined and I started to think about doing ome-xml storage but then lost momentum.

      External Email - Use Caution

carandraug, 2025-04-26:
Would be simpler if instead of having to implement the whole writing of ome-zarr as part of cockpit experiment, we have a new command that converts an existing dv file from cockpit into ome-zarr? Sure it's not ideal in terms of workflow but I think it will be a good starting point to test how useful will be to change from dv to ome-zarr and also hwo to use those libraries..

juliomateoslangerak, 2025-09-17:
Please, give it a spin. The changes are in my ```dev``` branch. I merged there the branch doing multiple channels (arbitrary interlacing of cameras) and it seems to work. 
I need to do some tests on hardware and with real samples. Also I want to see if there is any performance improvement as zarr is supposed to run asynchronously.