microscope-cockpit/cockpit #882: Add a modal dialog triggered if the experiment is truncated.

iandobbie, 2023-10-03:
The log window contains an error if the experiment is truncated as not enough data has arrived, either something went wrong,  maybe one or more images were dropped, or the users cancelled the collection. They have a couple of occasions that this happened at Diamond and the user was expecting the data to reconstruct. 

I suggest that we put up a modal dialog to say at least something went wrong and the data file is not complete.

carandraug, 2023-10-04:
Basically, we're forcing the user to acknowledge the error to avoid the error going unnoticed. Should we expand this to all errors (because if something escalates to the point of error, then it requires user attention)?

iandobbie, 2023-10-04:
Yes. Not sure if we should force this for every error. Worth considering but also might be very annoying for the user. 

This particular scenario should be rare and is important ensure the user knows that the collected data is likely useless.

iandobbie, 2023-10-18:
I added a modal dialog that say file truncated ok to contnue to cancel to delete file, but then does not yet delete the file. Code is in https://github.com/iandobbie/cockpit/tree/truncated-file-modal-diaglog