microscope-cockpit/cockpit #887: wx.timer polling power from my deepstar makes cockpit sluggish

juliomateoslangerak, 2023-10-27:
I used to have an issue on my deepstar reading the power. For some reason that I never identified it was taking time to get an answer. Tired of troubleshooting this, I just put a ```sleep(.1)``` and done.
However, now cockpit is very sluggish after last merges. I found that the gui is polling the power every 500 msec and that is calling that sleep. If I remove the sleep, things are not much better as there are errors thrown back.
As I see I have a 2 options.
- Try to figure out what the issue is with my Deepstar (I really checked the code and it is identical to master apart from the sleep). Anyone had a similar issue? I already tried connecting it through another (newer) PC to see if it had something to do with that.
- Not checking from the laser the power but just returning the set point (the laser does not have the APC option anyway). Do the Deepstars at Oxford have this APC? might that be a reason why mine is slower to return power?

Any other idea?

Maybe more of a microscope issue after all

juliomateoslangerak, 2023-10-27:
Actually I'm not sure of my interpretation. I made the deepstar return the power set-point without asking anything but the gui continues to be "shuttered".

juliomateoslangerak, 2023-11-03:
Let’s put a hold on this. It seems that this is not an issue when I start cockpit from the terminal. 
i have to look further into this.

iandobbie, 2023-11-09:
I cant remember if the Oxford deepstars had the ability to return power levels, but having it interrogated that often does seem a bit excessive. The idea is to have a valid number on the interface but probably once per s would be plenty, even then it might be better to think about doing it rapidly if you see or expect a chnage and then having less frequent polling.

juliomateoslangerak, 2023-12-06:
So I check on this and indeed running through the terminal is not sluggish cause it uses the installed cockpit version and not the development branch. So the issue is valid.

I'll try to change the polling to 1 per sec to see if this changes behavior.

juliomateoslangerak, 2023-12-08:
We may indeed reduce the frequency of polling. Still, it is worth considering why this is freezing the GUI. Are alternatives to this (eg, async). There might be some slow devices to answer or many devices.

iandobbie, 2023-12-11:
The current setup it pretty simplistic. I think a async call would be much more sensible. I will have a tink about how to set it up.