microscope-cockpit/cockpit #891: Cockpit once again reliably crashing on exit.

iandobbie, 2024-04-04:
On my mac cockpit now crashes regularly (maybe every time) on exit. It is a crash in thread 0, is a segv and appears to be trying to free some unallocated memory. 

I have upgraded to python 3.12.2 and not sure if this is a python upgrade issue or something in cockpit.

iandobbie, 2024-04-04:
Running from the command line give a freetype error so I think this might just be a new freetype install that needs fixing again with the python upgrade. I will investigate

carandraug, 2024-04-04:
I think there is no release of freetype with the fix we sent them. You may need to install freetype from dev sources.

carandraug, 2024-04-04:
Actually, here it is from you a few months ago :)

> Just to note that new installs on macOS still need to use the master branch of freetype-py from github rather than the current release from PyPI to avoid this crash roughly 1 time in 5.

_Originally posted by @iandobbie in https://github.com/microscope-cockpit/cockpit/issues/684#issuecomment-1888040947_ on 11 Jan, 2024

(but maybe there is a new bug?)

iandobbie, 2024-04-05:
Ok, installed the freetype from my src directory and appear to have stopped the freetype crash however the code still crashes every time I exit on 3.12, whereas exactly the same code closes cleanly on 3.11. 

On 3.12 I am using a venv because python complained about me installing it globally, which is probably actually brew complaining. I will try to run the same code under 3.11 in a venv.

The other strange thing is that when I run from the repl (eg python; import cockpit; cockpit.main(["cockpit'])). Cockpit closes cleaning but the crash happens when a close the python repl.

iandobbie, 2024-04-05:
A bit more info, when setup a separate venv with 3.11.8 it does not segfault. I ran it in a loop and of 20 repeats cockpit crashed once in a wxtimer so presumably there is still a lingering issue with one of the timers. 

In 3.12.2 it crashes every time, as in the output below (the 

```
  (cockpit_env) ID@IIC-ID-AIR cockpit % cockpit        
  FALLBACK (log once): Fallback to SW vertex for line stipple
  FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
  FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000
  zsh: segmentation fault  cockpit
```

(the FALLBACK lines are always there and I ought to work out a way to supress them) but it alwasy ends in a segfault, however when a loop it with a tiny shell script 
``` 
#!/bin/zsh
n=0
while [[ True ]];do
    n=$(($n+1))
    echo $n
    cockpit 2>&1 
done
```
it does not report the segfault. I guess the segfault is reported to the subshell that runs cockpit and that is just swallowed somewhere.

carandraug, 2024-04-09:
We have tracked this down to wxWidgets/Phoenix#2455 . The problem is limited to the wxPython 4.2.1 binary build for Python 3.12 and MacOS. The issue is already fixed in wxPython but no release has been made yet.

Possible workarounds:

-  install wxPython from [its snapshot-builds](https://wxpython.org/Phoenix/snapshot-builds/). Currently, the latest build is https://wxpython.org/Phoenix/snapshot-builds/wxPython-4.2.2a1.dev5670+a207b407-cp312-cp312-macosx_10_10_universal2.whl . Only the last 15 snapshots are kept so this link may no longer work in a couple of months. In that case, look for the most recent file with the `-cp312-cp312-macosx_10_10_universal2.whl` sufix.

- The issue appears to be on the sip version used to do the build which was pinned to `sip==6.7.9` (it has since been changed to be `sip==6.7.11` ). sip is not required at runtime, only for the build. So either:
  -  build wxPython from development sources
  - build wxPython 4.2.1from  source with a more recent version of sip

Anyway, it seems there is nothing we can do in Cockpit other than wait for wxPython to be re-released.

closed 2024-04-09 by carandraug