microscope-cockpit/cockpit #892: help in debugging configuration

carandraug, 2024-06-07:
A few weeks ago I spent a lot of time debugging an issue with the parsing of configuration files. I'm thinking it would be nice to have a window displaying the current configuration (which is the merge of multiple files). I have a proof of concept: it adds two entries on the menu bar for "View Cockpit Configuration" and "View Depot Configuration".

carandraug, 2024-06-07:
checkout my `configparser-frame` branch (new menu entries under "File").

iandobbie, 2024-06-08:
Seems to work well. Just a couple of thoughts.

1) The text window that appears on macos is tiny. I suggest we edit to have the following

    text_ctrl = wx.TextCtrl(
        panel,
        value=config_str.getvalue(),
        style=(wx.TE_MULTILINE|wx.TE_DONTWRAP|wx.TE_READONLY),size=(500,500)
    )

The added arbitary size gives a reasonable sized window that can be resized to be bigger if needed.

2) Can we add the current channels config to the list?

carandraug, 2024-06-08:
> The added arbitary size gives a reasonable sized window that can be resized to be bigger if needed.

Good point. But instead of an arbitrary size, I added the logic to resize the window to at least the size of the text or 80x24 characters.

> 2. Can we add the current channels config to the list?

Possibly. I wrote the code to display a configparser and I plan to add extra bits that show the list of files that were parsed and the list of files that were attempted. So it doesn't really work for channels configuration which is always a single file and in json format. It would require some extra work.

The goal was to help in figuring out what is the final configuration and where it is coming from. Is that an issue with channels too? (location of channels file is displayed on this new cockpit config window, and we have menu entries to import and save channels to specific locations).

iandobbie, 2024-06-10:
> > The added arbitary size gives a reasonable sized window that can be resized to be bigger if needed.
> 
> Good point. But instead of an arbitrary size, I added the logic to resize the window to at least the size of the text or 80x24 characters.

Much more elegant. Thanks for fixing my hack. 

> Possibly. I wrote the code to display a configparser and I plan to add extra bits that show the list of files that were parsed and the list of files that were attempted. So it doesn't really work [...]

No my main thought about this was to make sure we could find what was stored in the channel config. As far as I can tell the only way to do so at the moment is to export the channel and then go look at the text file. As this is already effectively displaying text files I thought it might be a simple way to add that functionality. Also I don't think it is a json file but a python dictionary.

carandraug, 2024-06-10:
> [...] Also I don't think it is a json file but a python dictionary.

It's json https://github.com/microscope-cockpit/cockpit/blob/master/cockpit/interfaces/channels.py#L151 (but they are very similar)