microscope-cockpit/cockpit #893: Python 3.12 doesn't like \ in strings.

iandobbie, 2024-06-08:
So we have a few regexps in the code that contain \ escape sequences that are now syntax errors. I think these just need to be replaced with raw strings. 

/Users/ID/src/cockpit/cockpit/devices/microscopeDevice.py:369: SyntaxWarning: invalid escape sequence '\s'
  self.cameras = re.split('[,;]\s*', cdefs)
/Users/ID/src/cockpit/cockpit/devices/microscopeDevice.py:376: SyntaxWarning: invalid escape sequence '\s'
  self.lights = re.split('[,;]\s*', ldefs)
/Users/ID/src/cockpit/cockpit/devices/microscopeDevice.py:386: SyntaxWarning: invalid escape sequence '\s'

carandraug, 2024-06-08:
You are right. Apparently, this has been happening since python 3.6 but it was a `DeprecationWarning`. Python 3.12 bumped it to `SyntaxWarning` with the goal of bumping it to `SyntaxError` later.  See [python 3.12 changes](https://docs.python.org/3/whatsnew/3.12.html#other-language-changes):

> * A backslash-character pair that is not a valid escape sequence now generates a [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning), instead of [DeprecationWarning](https://docs.python.org/3/library/exceptions.html#DeprecationWarning). For example, re.compile("\d+\.\d+") now emits a [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning) ("\d" is an invalid escape sequence, use raw strings for regular expression: re.compile(r"\d+\.\d+")). In a future Python version, [SyntaxError](https://docs.python.org/3/library/exceptions.html#SyntaxError) will eventually be raised, instead of [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning). (Contributed by Victor Stinner in [gh-98401](https://github.com/python/cpython/issues/98401).)

iandobbie, 2024-06-08:
Simple fix to just make the regexps raw strings. Pushed above and closing as fixed.

closed 2024-06-08 by iandobbie

carandraug, 2024-08-16:
Seems like we missed some regexps that needed to be using raw strings. I just added them and also decided to use it on all regexps, even those that didn't need it yet.