microscope-cockpit/cockpit #898: sensitivity threshold on axis positions

LauraA-Octopus, 2024-08-19:
Hello, 
during a debugging session with @carandraug, we noticed that in the Macro Stage Viewer, the flickering positions of the three axis cause the arrow on the xy stage representation to change constantly as if it was constantly moving, while in reality it is not. I was wondering if it would be possible to add a feature to choose the precision level of the positions to avoid seeing the flickering and have a more static view of the stage.

juliomateoslangerak, 2024-08-19:
Which stage is this?
I guess there would be a possible setting for this parameter. For my PI stage this setting is there, but not implemented in microscope, yet.

LauraA-Octopus, 2024-08-19:
The stage I am using is an Attocube AMC300, this is implemented in my forked microscope repo and working. I haven't opened a merge request yet for it, though

juliomateoslangerak, 2024-08-19:
Ah, I think I didn't get your point right. What you would like is a sort of threshold below which the Macrostage window will not display displacement. Am I right?
I presume your stage is applying little constant corrections that are passed on to cockpit. 
Maybe it would it make more sense to implement this on the microscope side? lower the precision with which actual position is given...

LauraA-Octopus, 2024-08-19:
yes sorry, you got my point right! I think it would make sense to lower the position precision on the microscope side yes. Is this feasible?

carandraug, 2024-08-19:
I will add some clarification of the issue and the requested feature.

Cockpit is now constantly polling the stage position (1 second interval) (see also #897). @LauraA-Octopus 's stage is in closed loop mode. Each time Cockpit get its position it shows it has moved by a few nanometers back and forth. The visual effect of this small moves is that the MacroStage is constantly showing arrows about the stage having moved.

You can simulate this behaviour by having a small random value to the simulated stage position, like so:

```
diff --git a/microscope/simulators/__init__.py b/microscope/simulators/__init__.py
index 74b62a7..5a03c41 100644
--- a/microscope/simulators/__init__.py
+++ b/microscope/simulators/__init__.py
@@ -421,7 +421,7 @@ class SimulatedStageAxis(microscope.abc.StageAxis):
 
     @property
     def position(self) -> float:
-        return self._position
+        return self._position + random.random()
 
     @property
     def limits(self) -> microscope.AxisLimits:
```

If you then start cockpit with simulated devices and look at the macro stage, you will see arrows coming out of the stage position all the time.

The feature being requested in this issue is to have the macro stage ignore movements below a certain resolution.

iandobbie, 2024-08-19:
It sounds like having a settable parameter to define this accuracy would be useful. I think this hasn't been an issue as cockpit didn't have the polling behavior but I added it since my test stage is a ludl that has a controller with a joystick and the cockpit interface wasn't reflecting any movements until the stage pos was queired for some other reason.

juliomateoslangerak, 2024-08-19:
This parameter would be a microscope-stage parameter or a cockpit parameter?

carandraug, 2024-08-19:
> This parameter would be a microscope-stage parameter or a cockpit parameter?

I think this should be a cockpit parameter. The stage is still moving and cockpit still knows that it moved. What this parameter controls is only whether the macro stage window reports the stage as moving.

iandobbie, 2024-08-19:
This is a gui function so it seems like squashing the graphics updates are the sensible solution, so a cockpit side parameter.

iandobbie, 2024-08-19:
There is already a a parameter set in cockoit/gui/macrostage/macroStageBase.py around line 68

## Don't bother showing a movement arrow for
# movements smaller than this.
MIN_DELTA_TO_DISPLAY = .01

So As the stage pos is in um this is 10 nm, way too small. We shoudl set this to a more sensible default (100 nm? more?) but also allow it to be a config parameter so set from a save into config.py.

LauraA-Octopus, 2024-08-20:
Is it possible to tune the value from the cockpit interface? 
If that is 10nm maybe it is too small

carandraug, 2024-08-20:
> Is it possible to tune the value from the cockpit interface?

At the moment there is no interface to do it. But if you start the python console from inside cockpit (Ctrl+P), you can:

```
import cockpit.gui.macroStage.macroStageXY
xy = [x for x in wx.Window.FindWindowByName("Macro Stage").GetChildren()[0].GetChildren() if isinstance(x, cockpit.gui.macroStage.macroStageXY.MacroStageXY)][0]
xy._min_delta_to_display = 1000  # experiment changing this to something that works for you
```

iandobbie, 2024-08-20:
When you have a sensible value you can set it in your config.conf, the location of which is OS dependent but can also be defined on the command line with cockpit --config-file x.conf

LauraA-Octopus, 2024-08-21:
this sounds great, thank you all very much!

closed 2024-08-21 by LauraA-Octopus