microscope-cockpit/cockpit #900: Weird missing numpy error on windows.

iandobbie, 2024-08-20:
I have an error on cockpit startup but only on windows. Even with a "--no-config-file" flag the system produces the following error. 

```
Traceback (most recent call last):
  File "C:\Users\idobbie1\src\cockpit\cockpit\__init__.py", line 163, in OnInit
    cockpit.util.userConfig.initialize(self.Config)
  File "C:\Users\idobbie1\src\cockpit\cockpit\util\userConfig.py", line 109, in initialize
    _config = _loadConfig(_config_path)
  File "C:\Users\idobbie1\src\cockpit\cockpit\util\userConfig.py", line 45, in _loadConfig
    config = eval(fh.read())
  File "<string>", line 407, in <module>
NameError: name 'np' is not defined
```

I don't understand what is trying to call np where it isn't defined.

This is with a fresh clone of upstream/master. The same code appears to be working fine on my mac, with no startup troubles.

iandobbie, 2024-08-20:
Found the issue. It is in the user config which is of course system and user specific. The PC has...

```
'experimentAltitudes': [99.04205228121448,
                         99.04205228121448,
                         100.66569248254586,
                         100.66569248254586,
 .....
                         29.22552362396493,
                         np.float64(1250.0),
                         np.float64(1250.0),
                         np.float64(1250.0),
                         np.float64(1250.0)],
```

I was testing with numpy 2.0.0 just before I went away. I assume that the experiment altitudes need to be forced into a string before being saved in order to prevent this.

carandraug, 2024-08-20:
> My next question is why does --no-config-files still load the user config? Should we change the behaviour so it isn't loaded with this flag? 

See #577 . Yes, this is confusing but I'm not sure how to go about fixing this.

iandobbie, 2024-08-20:
Ok, good point. As you say not sure what the best thing about loading the config.py is. 

As to fixing the issue it doesn't seem to be trivial.
```
>>> import numpy as np
>>> np.__version__
'2.0.0'
>>> f=np.float64(15.34)
>>> f
np.float64(15.34)
>>> print(f)
15.34
>>> import pprint
>>> p=pprint.PrettyPrinter()
>>> p.pformat(f)
'np.float64(15.34)'
```

The config saving routine uses pprint.PrettyPrinter.pformat to format the output.

iandobbie, 2024-08-20:
I guess we could cast the value to a python float when adding to the list which seems to be handled properly by the PrettyPrinter routines.

carandraug, 2024-08-20:
Yes. From looking at it, I think something like this:

    ## Ensure that the value is a builtin float and not something else (see #900)       
    self.experimentAltitudes.append(float(self.curStagePosition[2]))
    cockpit.util.userConfig.setValue('experimentAltitudes', self.experimentAltitudes)

But did `curStagePosition` change type? Why is it a numpy float now?

iandobbie, 2024-08-20:
Exactly my fix, just tested and it works. 

It might have always been a numpy float but with numpy2 the prettyprint routines don't know how to print it.

iandobbie, 2024-08-20:
In macroStageBase.py we have
```
        ## As above, but for the current position.
        self.curStagePosition = numpy.zeros(3)
```

So yes it always seemed to have been a numpy array.

iandobbie, 2025-05-22:
Unintentionally this was pulled when I was fixing another issue. As the fix I made was pulled into the main branch I'm closing this issue.

closed 2025-05-22 by iandobbie