microscope-cockpit/cockpit #901: UI to set individual device settings

carandraug, 2024-08-21:
Currently, the only devices that expose the settings dict in cockpit are cameras through their settings button on the main window:

[image]

But all Python-Microscope devices have a settings dict. Would be nice to expose that but not all devices have space for them on the main window. I'm thinking the simplest thing would be to add a menu bar for devices that would open those as separate windows.  (requested by @LauraA-Octopus on image.sc https://forum.image.sc/t/adding-settings-on-stage-controller/100713)

carandraug, 2024-08-21:
I have implemented a minimal solution on my `901-devices-menu` branch:

[image]

This does not filter devices that are not Python-Microscope devices. We probably could group them by type somehow. And the "setings" button beneath the camera does some extra things that I'm not sure what they are for. But this is a start for people to comment and maybe use as hint to implement better.

iandobbie, 2024-08-21:
Seems like a good idea.

LauraA-Octopus, 2024-08-21:
This seems indeed a good idea, I can try it in the lab probably this Friday. Hopefully this will solve the issue, otherwise I will have to write a custom class for the AMC in cockpit, which I'd rather avoid since you are trying to be not device specific

juliomateoslangerak, 2024-08-21:
Good idea. I might consider renaming the menu to Settings or Device settings though.