microscope-cockpit/cockpit #910: MacOS does not accept wx.Menu items with id=0

juliomateoslangerak, 2025-02-26:
MacOS complains when an wx.Menu is created with id = 0. 
I noticed this while developing using simulated SLM on my Mac and is, I believe only a problem with cockpit/gui/device/Menu which implements a rather convoluted way of adding items to a menu. I think this could be simplified with a list of (label, callback) tuples and letting wx adding the menu ids.

Is there a reason for not doing so? 
If I'm not wrong, this is only used by SLM GUI.

juliomateoslangerak, 2025-02-26:
For example the mainWindo.ChannelsMenu uses a much simpler menu construction loop:
```python
        for label, method in control_items:  # a list of (str. callbacks)
            menu_item = self.Append(wx.ID_ANY, item=label)
            self.Bind(wx.EVT_MENU, method, menu_item)
```

iandobbie, 2025-02-26:
Seems like replacing the existing code with example above which is a pattern common elsewhere in the codebase seems like a better long term fix. The i+1 code in the temp fix just feels to me like it is going to be broken at some future date.

carandraug, 2025-02-26:
Julio Mateos Langerak writes:
> [...]
> Is there a reason for not doing so?
>
> If I'm not wrong, this is only used by SLM GUI.

I don't know why it is implemented in that manner.  At first glance, it
appears that we could change it to avoid passing the `id` argument.
Since this is only used in the SLM, can you test with the SLM?

Something like the `mainWindow.ChannelsMenu` example you show would fix
this with the least amount of code changes.  But if this is only needed
by boulderSLM, I think it shows that we don't actually need a "general"
`cockpit.gui.devices.Menu` class.  If that is true, then maybe implement
something simpler directly on boulderSLM.

carandraug, 2025-02-26:
David Miguel Susano Pinto ***@***.***> writes:
> Julio Mateos Langerak writes:
>
> I don't know why it is implemented in that manner.  At first glance, it
> appears that we could change it to avoid passing the `id` argument.
> Since this is only used in [...]

I forgot we have a DummySLM in microscope to test.

I have fixed this by dropping the whole Menu class which is used nowhere
else and made things more convoluted in the BoulderSLM class (because it
needed an extra layer to map index of button/ID of menu item to the
actual method).

juliomateoslangerak, 2025-02-26:
In fact the same is true for most of the wx helper classes in that module. They only serve a very limited number of non microscope devices (sim, linkcam, clarity,...). Not the case for a few (likely more recent) EnableButton and SettingsEditor (and TupleOfInts used by this later).
I guess this Menu class (and presumably the other ones) fall a bit short to provide significant simplification when developing a new device. Ideally we would like better integration with microscope devices but I don't think it is worth at this point to try to generalise.
I'm creating a sim_slm device that handles the sim logic and interfaces with a microscope.slm and I started from the existing sim code. Does it makes sense to review the gui code to not depend on those older classes and use EnableButton and SettingsEditor or create my own ones?

carandraug, 2025-02-27:
> Does it makes sense to review the gui code to not depend on those older classes and use EnableButton and SettingsEditor or create my own ones?

It's your call. Use them if you fell that they make the code simpler (and simpler is not only a function of the number of lines). Personally, I've been favouring use wx instead of relying on those helper functions. For most things, wx already has builtin helper functions and it's a matter of finding them. But for a few things, such as `EnableButton` and the `SettingsEditor` really is very specific so it makes sense to use them.