microscope-cockpit/cockpit #912: Ignoring images to save

juliomateoslangerak, 2025-04-26:
I'm looking through the dataSaver code and there is some functionality there to ignore images to be saved. I traced that to this method in the Experiment class `resetCams` that mentions:
```
    ## Given a set of cameras and a time, trigger the cameras and record that
    # we want to throw away the resulting image. This blanks the camera
    # sensors so they don't record light that we don't care about.
```
It is not clear to me in which scenario this is needed.

juliomateoslangerak, 2025-04-26:
I saw those comments too but, I don't quite get it. Are cameras that are not exposing subject to collect photons visible in their next exposure?

iandobbie, 2025-04-26:
Yes frame transfer cameras will always acquire data no mater what state they are in, until you explicity read out or wipe the image.

juliomateoslangerak, 2025-07-08:
I see.
I'm closing this issue.