microscope-cockpit/cockpit #913: SimulatedLightSource with no HW trigger produces errors in cockpit experiments.

iandobbie, 2025-05-09:
If you setup a microscope provided SimulatedLightSource without a HW trigger, ie has no triggerSource and triggerLine entries in the depot file then experiments fail in a very opaque manner. The error is below

```
Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Users\idobbie1\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Users\idobbie1\AppData\Local\Programs\Python\Python310\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\idobbie1\src\cockpit\cockpit\experiment\experiment.py", line 405, in execute
    events.executeAndWaitFor(events.EXPERIMENT_EXECUTION,
  File "C:\Users\idobbie1\src\cockpit\cockpit\events.py", line 308, in executeAndWaitFor
    return executeAndWaitForOrTimeout(eventType, func, None, *args, **kwargs)
  File "C:\Users\idobbie1\src\cockpit\cockpit\events.py", line 342, in executeAndWaitForOrTimeout
    func(*args, **kwargs)
  File "C:\Users\idobbie1\src\cockpit\cockpit\handlers\executor.py", line 209, in executeTable
    return self.callbacks['executeTable'](actions, 0, len(actions), numReps,
  File "C:\Users\idobbie1\src\cockpit\cockpit\devices\executorDevices.py", line 204, in executeTable
    self.connection.PrepareActions(actions, numReps)
  File "C:\Users\idobbie1\AppData\Local\Programs\Python\Python310\lib\site-packages\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Users\idobbie1\AppData\Local\Programs\Python\Python310\lib\site-packages\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
IndexError: list index out of range
```

iandobbie, 2025-05-09:
Previously I thought that this kind of config issue produced an error that there was no executor that could control the light source. This error was also pretty opaque but as least described where the issue was. The list index out of range error here is totally uninformative.

juliomateoslangerak, 2026-10-09:
I'm looking into this error in the context of a simulated environment with a stage aware camera. But the error I get is different. When there is no executor the lasers cannot handle the on-off actions. I have a fix in this commit 084fc0b but I'm not sure it is the same thing.