microscope-cockpit/cockpit #917: Running out of memory and crashing during large image acquisition.

iandobbie, 2025-06-09:
So this is the continuing story of large time lapse acquisition for SMLM. With the fixes discussed in #916 we can start the acquisition of 100k time points of a single plane. I ran with my raspberry Pi cam setup and the system acquired almost half the data at about 49k images but then crashed as it ran out of memory. The actual error came  in the viewCanvas code generating the histogram but I think that the cause is the fact that each acquired image is sitting somewhere in memory. A quick test shows the memory usage growing at about 1MB per image with an image of 1296x836 at 1 bits per pixel, so seems to be roughly the right size of just having the image left in memory somewhere.

Will investigate.

iandobbie, 2025-06-10:
So it seems this is the issue that Julio previously encountered as described in #775 and appears to be fixed by patch https://github.com/microscope-cockpit/cockpit/commit/2ce0ab2f1386de33811e5ba0e37d366734d7b687 

I just ran 10k frames without the memory required blowing up but for a true test UI am re-running the 100k frames to ensure that this will actually work.

juliomateoslangerak, 2025-09-05:
The system does not crash at all.
After some research I figured that this is caused by calls to setView by the SYNCED_VIEW event. When this event is called the ViewCanvas (self) is already gone, as you mention, presumably destroyed already. Making the following change (checking on self's existence) to setView seems to fix things:
from
```python
    def setView(self, panX, panY, zoom):
        self.panX = panX
        self.panY = panY
        self.zoom = zoom
        self.Refresh(0)
```
to
```python
    def setView(self, panX, panY, zoom):
        self.panX = panX
        self.panY = panY
        self.zoom = zoom
        if self:
            self.Refresh(0)
```
There are more calls to Refresh along the ViewCanvas. I will try to see if they also trigger the error